visrealm/pico9918 #12: Comments in main.c are misleading vis a vis CD0-CD7

thorpej, 2024-09-01:
The comments in main.c about the CD0-CD7 pin mappings are misleading.  TI's convention calls the LSB "CD7" and the MSB "CD0".  The nomenclature is correct on the v0.3 schematic, but the comments in main.c are incorrect.  A corrected comment block would be:

```
  * Pin  | GPIO | Name      | TMS9918A Pin
  * -----+------+-----------+-------------
  *  19  |  14  |  CD7      |  17         LEAST SIGNIFICANT BIT
  *  20  |  15  |  CD6      |  18
  *  21  |  16  |  CD5      |  19
  *  22  |  17  |  CD4      |  20
  *  24  |  18  |  CD3      |  21
  *  25  |  19  |  CD2      |  22
  *  26  |  20  |  CD1      |  23
  *  27  |  21  |  CD0      |  24         MOST SIGNIFICANT BIT
.
.
.
```

I also suggest renaming the constant ```GPIO_CD0``` to ```GPIO_CD7```.

visrealm, 2024-09-01:
Thanks. You're right, this was left over from v0.2 which actually had this pinout. They were reversed like this resulting in any gpio operations on the CD0-7 pins needing to be reversed via a lookup table. It was a messy time. 😁

visrealm, 2024-09-01:
Closed as fixed in dev

closed 2024-09-01 by visrealm