visrealm/pico9918 #17: Compiling from command line errors

chris-jh, 2024-10-13:
Two errors show up when trying to build from command ( i am on a mac doing this, but think it would be same for all OS's)

First was to do with panic.c and second was to do with irq.c from the pico-idk.

panic.c was 'error: expected ')' before ':' token"

ira.c was "fatal error: pico/runtime_init.h: No such file or directory"

i fixed both errors by changing the /src/pio-utils/CMakeLists.txt file

I added

`target_compile_options(${PROJECT_NAME} PUBLIC -include pico/platform/compiler.h)`

at the end of the file

and changed this line to include the pico_stdlib

`target_link_libraries(${LIBRARY} PUBLIC hardware_pio pico_stdlib)`

For me this made it build, (after installing pillow for python also)

I built a v0.3 pcb, what do i change to build for this PCB ?

visrealm, 2024-10-14:
Thanks for the report. I don't have a Mac to test, but if your changes work on my other test platforms, I'll add them in.

Building for 0.3 is just a matter of setting:

`set(PICO9918_PCB_MINOR_VER 3)`

in src/CMakeLists.txt

chris-jh, 2024-10-14:
Thought it was this, just that the uf2 file still had 0.4 in the name and wasn't 100% sure.

One more thing i noticed with the 0.3pcb, not sure about your 0.4. I can see the cpuclock on pin 23 on the pico itself. but what comes out to the main board, it is not a clock signal.

I assume the HC08 is doing something to the signal. i also have 0 ohm resistors installed.

Working on a clone SVI-328 and had the RGB output working (TMS9929ANL), but then i broke my 3.54Mhz on the main board, so thought could use the VDP CPU Clock and that is when i noticed the signal was not correct.

I can add this an other issue if you like

Will this output a composite signal at some point ?

visrealm, 2024-10-15:
I've checked a couple of my v0.3 boards on the scope and verify, I'm getting a 3.579MHz signal out of the CPUCLK pin of the TMS9918A socket on both of them. There must be something awry between the PICO pin 23 and the TMS9918A pin as you said.  The HC08 just boosts the 3.3V signal to 5V having the 3.3V signal going to both inputs of one channel and the output going to the CPUCLK 0Ohm resistor/jumper.

I have no plans to support composite output. I think it's technically possible, but it's not something I'm interested in.

visrealm, 2024-11-05:
I've added your changes. Closing as fixed.

Thanks.

closed 2024-11-05 by visrealm