visrealm/pico9918 #20: Pico RP2040 black clone cannot boot.

repetto74, 2024-11-01:
Hello again :-)

Any idea why I cannot get this Pico black clone to show up on my PC? I am under MacOSX but with either the purple clone or the original Pico I can get them to show-up as disk drives on my desktop once connected in boot mode.

For this black Pico nothing is showing up when I hook it up to my PC with the boot button pressed :-( 

[image]

repetto74, 2024-11-01:
The pico is busted :-(. I have another black Pico and this one works. Too bad I have soldered the one not working!

visrealm, 2024-11-01:
That's a great point and probably is the cause of these issues. I suggest removing the RST jumper. You don't really need it.

repetto74, 2024-11-01:
Thanks I will try to remove the 0 ohm resistor on the RST position and test again.

repetto74, 2024-11-02:
I confirm that removing the RST jumper will fix this issue and the pico will boot also if installed on the PCB :-).

repetto74, 2024-11-03:
So all looks good on my MSX HX-10. The wrong purple tint is probably due to missing 75 ohms resistors in my SCART cable. I have to add those resistors on the output connector in the MSX. Looks good anyway! :-D 
[image]

visrealm, 2024-11-04:
From AA:

4 hours ago, [visrealm](https://forums.atariage.com/?app=core&module=members&controller=profile&id=77331) said:
This looks like the green output of RGB isn't connected.

Ok, I'll need to check the wirings then.😅 Maybe a soldering not properly done. I'll check with the schematic.

EDIT: I found that resistor array RN6 has two legs not properly soldered. There is no continuity between GREEN signal and G1.

Closing

closed 2024-11-04 by visrealm