visrealm/pico9918 #23: Is expanded functionality planned?

Unilein, 2024-11-08:
Is there a plan to expand the functionality in principle? For example, an expanded or alternative color palette? Or other resolutions, more sprites or multi-colored sprites?

I use a classic TMS9918A in my homebrew project and am considering switching. Either to a self-developed unit or to your project, if there are any further extensions to be expected.

Regards
Unilein

visrealm, 2024-11-08:
Hi Unilein.

There is already F18A support. The F18A has custom palettes, 80 column mode, multi-color sprites, multi-color tiles, hardware scrolling, a bitmap layer plus two tile layers. Here's a quich demo video showing the PICO9918's F18A compatibility mode in action: [PICO9918 F18A mode preview demo (screen capture)](https://youtu.be/TabTIPL1xQY).

F18A mode isn't yet in the main branch, but I have binaries for it: https://github.com/visrealm/pico9918/releases/tag/v0.4.1-f18a-preview1

I'm also working on V9938 compatibility.

Cheers
Troy

Unilein, 2024-11-09:
Great!

I will order PCB's today :-)

Cheers
Unilein

closed 2024-11-09 by Unilein