visrealm/pico9918 #26: "Issue with CSW_L_PIN in Pico9918 VRAM Write Operation Affecting Color Cycle and Text Display

DonaldMoran, 2024-12-05:
I have v0.4.3, on a pico from DWEII, the USB-C one, which I intend to use on a homegrown 65c02. To learn to use it, I have first wired it up to a second Pico W, with the intention of cycling through the colors and writing "Hello, World!" to the screen.

The following code cycles the colors as expected; however, I had to comment out the lines in void write_vram that set the CSW_L_PIN low, i.e., //gpio_put(CSW_L_PIN, 0);

If I uncomment those lines, the first screen color change occurs, but from there, no additional screen color changes and no text appears. The program does run to completion.

I feel I might be doing something wrong here, but reaching out as my research hasn't helped thus far. I am new to the Pico9918 and the TMS9918A in general.

If you could take a quick look, I would appreciate it. Also, knowing this uses the emulation library, I have assumed there is a default character pattern.

PS - Nice seeing you over at the 6502 forum and appreciate you.

```
#include "pico/stdlib.h"
#include <stdio.h>

#define CSW_L_PIN 27
#define MODE_PIN 28
#define DATA_PINS {14, 15, 16, 17, 18, 19, 20, 21}  // CD7 (bit 7) to CD0 (bit 0)

const char* color_names[] = {"TRANSPARENT", "BLACK", "MEDIUM GREEN", "LIGHT GREEN", "DARK BLUE", "LIGHT BLUE", "DARK RED", "CYAN", "MEDIUM RED", "LIGHT RED", "DARK YELLOW", "LIGHT YELLOW", "DARK GREEN", "MAGENTA", "GRAY", "WHITE"};
const uint8_t ti_colors[] = {0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F};

void setup_gpio();
void set_data_pins(uint8_t byte);
void send_byte(uint8_t byte);
void set_mode();
void write_vram(uint16_t address, uint8_t data);
void write_name_table(uint16_t address, uint8_t data);
void display_text();
void cycle_colors();

int main() {
    stdio_init_all();
    printf("Starting setup...\n");

    // Initial wait
    printf("Initial wait...\n");
    sleep_ms(5000);

    // Double initialization to stabilize state
    printf("First initialization...\n");
    setup_gpio();
    set_mode();
    
    printf("Second initialization...\n");
    setup_gpio();
    set_mode();

    // Cycle through TI-99/4A TMS9918A colors with "AB" text
    cycle_colors();

    printf("Program complete.\n");

    return 0;
}

void setup_gpio() {
    static const uint8_t data_pins[] = DATA_PINS;

    for (int i = 0; i < 8; i++) {
        gpio_init(data_pins[i]);
        gpio_set_dir(data_pins[i], GPIO_OUT);
        gpio_put(data_pins[i], 0);  // Set all data pins low initially
    }
    gpio_init(CSW_L_PIN);
    gpio_set_dir(CSW_L_PIN, GPIO_OUT);
    gpio_put(CSW_L_PIN, 1);  // Set CSW_L_PIN high initially

    gpio_init(MODE_PIN);
    gpio_set_dir(MODE_PIN, GPIO_OUT);
    gpio_put(MODE_PIN, 1);  // Set MODE_PIN high initially (command mode)

    stdio_init_all();
    printf("GPIO setup complete.\n");
}

void set_data_pins(uint8_t byte) {
    static const uint8_t data_pins[] = DATA_PINS;
    for (int i = 0; i < 8; i++) {
        gpio_put(data_pins[i], (byte >> i) & 1);  // Set data pins
    }
    sleep_us(50);  // Small delay to stabilize
    printf("Data pins set to 0x%02X\n", byte);
}

void send_byte(uint8_t byte) {
    set_data_pins(byte);
    gpio_put(CSW_L_PIN, 0);  // Start the write operation
    sleep_us(200);  // Delay to ensure the byte is latched
    gpio_put(CSW_L_PIN, 1);  // End the write operation
    sleep_us(100);  // Small delay to stabilize
    printf("Data byte 0x%02X sent.\n", byte);
}

void set_mode() {
    printf("Setting mode to Text mode (000 1011)...\n");
    send_byte(0x80 | 0x00);  // Register 0 (Mode)
    send_byte(0x0B);  // Text mode (000 1011)
    printf("Mode set to Text mode.\n");
}

void write_vram(uint16_t address, uint8_t data) {
    printf("Preparing to write VRAM - Address: 0x%04X, Data: 0x%02X\n", address, data);

    gpio_put(MODE_PIN, 0);  // Switch to VRAM mode
    sleep_us(50);  // Small delay to stabilize

    // High byte of address
    set_data_pins(0x40 | (address >> 8));
    //gpio_put(CSW_L_PIN, 0);  // Start the write operation
    sleep_us(50);  // Small delay to stabilize
    gpio_put(CSW_L_PIN, 1);  // End the write operation
    sleep_us(50);  // Small delay to stabilize

    // Low byte of address
    set_data_pins(address & 0xFF);
    //gpio_put(CSW_L_PIN, 0);  // Start the write operation
    sleep_us(50);  // Small delay to stabilize
    gpio_put(CSW_L_PIN, 1);  // End the write operation
    sleep_us(50);  // Small delay to stabilize

    // Data byte
    set_data_pins(data);
    //gpio_put(CSW_L_PIN, 0);  // Start the write operation
    sleep_us(50);  // Small delay to stabilize
    gpio_put(CSW_L_PIN, 1);  // End the write operation
    sleep_us(50);  // Small delay to stabilize

    gpio_put(MODE_PIN, 1);  // Switch back to command mode
    sleep_us(50);  // Small delay to stabilize

    printf("Finished writing to VRAM - Address: 0x%04X, Data: 0x%02X\n", address, data);
}

void write_name_table(uint16_t address, uint8_t data) {
    write_vram(address, data);
}

void display_text() {
    const char* text = "AB";
    uint16_t vram_address = 0x3800;  // Starting VRAM address for name table
    while (*text) {
        printf("Displaying character: %c at VRAM address: 0x%04X\n", *text, vram_address);
        write_name_table(vram_address++, *text++);
        sleep_us(200);  // Small delay to stabilize
    }
}

void cycle_colors() {
    printf("Cycling through TI-99/4A TMS9918A colors...\n");
    for (int i = 0; i < 16; i++) {
        send_byte(0x80 | 0x07);  // Write to VDP register 7 (Background color)
        send_byte(ti_colors[i]); // Set the background color
        printf("Displaying color %s (0x%02X).\n", color_names[i], ti_colors[i]);
        sleep_ms(2000);  // Delay to display each color

        // Display text on each color screen
        display_text();
    }
    printf("Color cycling complete.\n");
}
```

visrealm, 2024-12-06:
Hi, you definitely need those lines of code. Without them, you're not getting the required pulses on the CSW pin.

1. Is your PICO9918 CSR pin held high? It should be.
2. There are no default patterns in its VRAM. It emulates a real TMS9918A which obviously also doesn't have any default data in VRAM.

So, you'll need to check the CSR line. And also, you'll need to load font pattern data.

DonaldMoran, 2024-12-06:
I appreciate your response. Yes, I have tied CSR high to 3.3V. I will look into loading the font pattern data as suggested, however this strange issue continues with vram_write. When I uncomment any of the lines that bring CSW_L_PIN low within the write_vram routine, the screen changes stop. Interestingly, the set of CSW_L_PIN low is within the send_byte() while cycling the screen colors without any issues. Initially, I used send_byte() for VRAM writes, but I separated it out while troubleshooting this.

visrealm, 2024-12-06:
Just looking at your write_vram() function. The address needs to be written with MODE high.

DonaldMoran, 2024-12-06:
Thank you, that will help for sure, however I do still get the same result. Also visrealm,  I realize this will surely be an issue from my lack of knowledge and not a repo issue, I just thought this may be the best way to reach out. I will keep working at this until I get it! I know of the examples and data sheets. If you know any other good resources for a beginner like me, I would appreciate being pointed in that direction. I am familiar with the pico as well as pio, it is the tms9918a I am new to. I love this project by the way, awesome work! This is something to be proud of for sure! Thanks for your time.

DonaldMoran, 2024-12-06:
Hello visrealm,

I've updated my code to load a pattern into the pattern table, making sure the mode pin is set high for writing addresses and low for data. I've reverted to using the send_byte() routine and ensured that the proper pin for CSR is tied high.

The code below runs and cycles through the screen colors. However, it currently only does so if the following lines are commented out, or set to prevent any data from being sent to the pico9918:

In the int main(), I have commented out the loading of the font pattern.
```    // Load the font pattern for 'A' into VRAM
    //load_font_patterns();
```

In the display_text() function, I have set the single character that I am attempting to write to the screen as null.

```
     //const char* text = "A";
    const char* text = "";
```

The code provided will facilitate changes in screen color.

```#include "pico/stdlib.h"
#include <stdio.h>

#define CSW_L_PIN 27
#define MODE_PIN 28
#define DATA_PINS {14, 15, 16, 17, 18, 19, 20, 21}  // CD7 (bit 7) to CD0 (bit 0)

const char* color_names[] = {"TRANSPARENT", "BLACK", "MEDIUM GREEN", "LIGHT GREEN", "DARK BLUE", "LIGHT BLUE", "DARK RED", "CYAN", "MEDIUM RED", "LIGHT RED", "DARK YELLOW", "LIGHT YELLOW", "DARK GREEN", "MAGENTA", "GRAY", "WHITE"};
const uint8_t ti_colors[] = {0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F};

//#define TMS_DEFAULT_VRAM_NAME_ADDRESS          0x3800
//#define TMS_DEFAULT_VRAM_COLOR_ADDRESS         0x0000
//#define TMS_DEFAULT_VRAM_PATT_ADDRESS          0x2000
//#define TMS_DEFAULT_VRAM_SPRITE_ATTR_ADDRESS   0x3B00
//#define TMS_DEFAULT_VRAM_SPRITE_PATT_ADDRESS   0x1800

// Font pattern for the character 'A'
const uint8_t font_patterns[] = {
    0x18, 0x24, 0x42, 0x42, 0x7E, 0x42, 0x42, 0x42
};

void setup_gpio();
void set_data_pins(uint8_t byte);
void send_byte(uint8_t byte);
void set_mode();
void write_vram(uint16_t address, uint8_t data);
void write_name_table(uint16_t address, uint8_t data);
void load_font_patterns();
void display_text();
void cycle_colors();

int main() {
    stdio_init_all();
    printf("Starting setup...\n");

    // Initial wait
    printf("Initial wait...\n");
    sleep_ms(5000);

    // Double initialization to stabilize state
    printf("First initialization...\n");
    setup_gpio();
    set_mode();
    
    printf("Second initialization...\n");
    setup_gpio();
    set_mode();

    // Load the font pattern for 'A' into VRAM
    //load_font_patterns();
[... 94 lines cut ...]
    //const char* text = "A";
    const char* text = "";
    uint16_t vram_address = 0x3800;  // Starting VRAM address for name table
    while (*text) {
        printf("Displaying character: %c at VRAM address: 0x%04X\n", *text, vram_address);
        write_name_table(vram_address++, *text++);
        sleep_us(200);  // Small delay to stabilize
    }
}

void cycle_colors() {
    printf("Cycling through TI-99/4A TMS9918A colors...\n");
    for (int i = 0; i < 16; i++) {
        send_byte(0x80 | 0x07);  // Write to VDP register 7 (Background color)
        send_byte(ti_colors[i]); // Set the background color
        printf("Displaying color %s (0x%02X).\n", color_names[i], ti_colors[i]);
        sleep_ms(2000);  // Delay to display each color

        // Display text on each color screen
        display_text();
    }
    printf("Color cycling complete.\n");
}
```
I would appreciate any further guidance you can provide. Thank you for your time and effort.

DonaldMoran, 2024-12-08:
Hi visrealm,

I think I'm having trouble because I've been trying to do this without the PCB, just using a couple of pico boards. I suspect the logic of the other chips, like the 74hc08, is necessary for this to work correctly. Though I do have a TI-99/4A and intend to get the PCB, I was trying to make this work on a breadboard with a homebrew 6502, thinking the PCB wasn't needed as I didn't need GROMCLK or CPUCLK.

At least through trying to make this work, I have learned more about how to work with the pico9918/tms9918!

I should probably close this issue, as I suspect there isn't any issue with the repository. I just thought this was the best way to reach out to you. I'll leave it open for now in hopes of a reply.

Thanks for your awesome project!

DonaldMoran, 2024-12-14:
This is obviously not a repo issue, but most likely because I've been trying to do this without the PCB, just using a couple of pico boards. I did attempt to order the PCB, but was stalled with the question: "PCBWay needs clarification on whether your Pico9918 v0.3 board is designed as a 4-layer or 2-layer PCB."

Thanks.

closed 2024-12-14 by DonaldMoran

visrealm, 2024-12-16:
Hi, all boards are 4 layers.

I do plan to try your code once I get the time to set it up.

DonaldMoran, 2024-12-17:
Excellent, thank you visrealm! I'll look at JLCPCB. If you do get the chance to take a look at this, please post back, I'll be interested.

Merry Christmas to you and yours!