visrealm/pico9918 #27: Issue with colors with Extended Basic

JonVogel, 2025-01-05:
When I run a simple program to set screen and character colors in Ext Basic, it doesn’t work. The screen flashes and then the color goes back Ext Basic blue.  This does not occur with console basic. 

Example:

10 call clear
20 for c = 2 to 16
30 call screen(c)
40 call color(1, c, c)
50 for t = 1 to 500
60 next t
70 next c

visrealm, 2025-01-05:
Hi Jon,
I'm unable to reproduce this one. I've tried a couple of different versions of Extended BASIC (including genuine cart).
Any other details you might be able to provide?

JonVogel, 2025-01-05:
I ran a test with my PEB unplugged and it worked fine.  I have a HRD in my PEB and I’m thinking that there’s and interaction with it and Ext Basic thats causing this.

JonVogel, 2025-01-05:
With the HRD pulled from the PEB, the issue doesn’t occur.  It looks like the HRD DSR is putting the screen back to its menu color as soon as Ext Basic tries to set it to a different color. 

The problem also occurs with an unmodified console and the HRD, so this is clearly an HRD issue that I never noticed before.  Very strange.

closed 2025-01-05 by JonVogel

visrealm, 2025-01-06:
Thanks for investigating, Jon.
I don't have a PEB, so can't do any further testing, but glad you worked it out :)