visrealm/pico9918 #30: nocutr parts list

smccloud, 2025-01-28:
I'm looking at buying a nocut PCB from Arcadeshopper since shipping is much less than it is from you in Australia (sorry).  However, I'm wondering if you have a parts list for the JST version (including the cable).

MMarcoux66, 2025-01-28:
If you are in the US and can solder, I bought 5 of the boards and parts but only need 2 or 3 so I have extras. I have the JST version.

MMarcoux66, 2025-01-28:
Are you talking about the JST cable that goes from the PICO9918? One should come with the chip. 

I have the PCB board, both audio jacks, the JST connector, jumper and the VGA port. I have 3D printed the cover and the spacers. I just have not had time to do the soldering. I have enough to build 5. If you want to build your own, I can hook you up with the parts as I don't need 5 :).

Since I wanted to use one in a ColecoVision, Tony pointed me to an extension that you can get from Aliexpress. Is that the jumper cable you are looking for?

smccloud, 2025-01-28:
I am talking about the female-to-female JST cable; my PICO9918 came with a female JST-to-female VGA cable.  As for the 3D printed housing, I should have my "slightly modified" Ender 3 back up and =working again soon so I can print my own.

MMarcoux66, 2025-01-28:
Gotcha. I totally missed that when I made my shopping list. I need to look for some for myself. What you are looking for is JST 1.25mm cables. I will have to look more later, but these would work if spliced.

https://www.aliexpress.us/item/3256802350081389.html?spm=a2g0o.productlist.main.17.4f557945Nmxa60&algo_pvid=d585a0cd-b95b-4fce-8709-cb369eb47a9d&algo_exp_id=d585a0cd-b95b-4fce-8709-cb369eb47a9d-8&pdp_npi=4%40dis%21USD%210.95%210.74%21%21%210.95%210.74%21%402103244417381002094955852e885d%2112000021081608453%21sea%21US%211602307906%21X&curPageLogUid=imlN1HOCeO5k&utparam-url=scene%3Asearch%7Cquery_from%3A

smccloud, 2025-01-28:
Looks like these should work, I hate ordering from Aliexpress.
https://amzn.to/4gici0V

visrealm, 2025-01-28:
I'll do up a BOM for each board type in the next day or so. Just be aware, those JST cables from Amazon will need wither the connector soldered to the bottom of the PCB or the pins manually reversed on one side.

smccloud, 2025-01-28:
> I'll do up a BOM for each board type in the next day or so. Just be aware, those JST cables from Amazon will need wither the connector soldered to the bottom of the PCB or the pins manually reversed on one side.

I can reverse the pins; it is no huge deal for me.  I'm odd in that soldering through hole stuff is relaxing for me.  Uncloging my Ender 3 hotend right now, not so much......

MMarcoux66, 2025-01-28:
Does $10 plus shipping sound fair? I believe that I have $50 total in parts.

Yeah, those Enders can be a lot of fun. I bought a Bambu a year ago and love it compared to the Ender it replaced...

MMarcoux66, 2025-01-28:
Cool. Feel free to email me at mmarcoux66 on Hotmail or same ID through Atariage PM messenger. Whichever you prefer.

Yeah, that is some crazy stuff with the FW but I do love my Bambus and too late for me to change since I already made the investment.

visrealm, 2025-01-28:
It also just occurred to me, if you're ordering from ArcadeShopper, he has the latest version (v1.3) which can be with JST or FFC. It uses a slimmer VGA connector than the order style JST-only boards. If @MMarcoux66 has the previos JST-only board, the VGA connector won't fit.

This is the VGA connector for the combined no-cut board: https://www.lcsc.com/product-detail/D-Sub-VGA-Connectors_HOAUC-HYC06-HDR15B-060_C711364.html

MMarcoux66, 2025-01-28:
Yes, when I originally ordered parts, I ordered that slimmer version instead of the bigger one for the boards which I have.

visrealm, 2025-01-29:
> I'm looking at buying a nocut PCB from Arcadeshopper since shipping is much less than it is from you in Australia (sorry). However, I'm wondering if you have a parts list for the JST version (including the cable).

I've just noticed his PCB price is the same as my full kit price. $20USD. Shipping will be extra for sure, but probably wouldn't turn out to be any cheaper overall I don't think.

smccloud, 2025-01-30:
> It also just occurred to me, if you're ordering from ArcadeShopper, he has the latest version (v1.3) which can be either JST or FFC. It uses a slimmer VGA connector than the older style JST-only boards. If [@MMarcoux66](https://github.com/MMarcoux66) has the previous JST-only board, the VGA connector won't fit.
> 
> This is the VGA connector for the combined no-cut board: https://www.lcsc.com/product-detail/D-Sub-VGA-Connectors_HOAUC-HYC06-HDR15B-060_C711364.html

My PICO9918 has the VGA connector directly connected to the JST wires, so it doesn't matter to me either way.